cpb-/yocto-cooker #29: Error message after cooking

cpb-, 2020-03-30:
Everything compile fine, but an error message from `chef` appears at the end:

```
[Yocto-trees@why-cpb]$ chef cook hytem-systems.menu 
Update layers in project directory
Updating source /home/testing/Build/Yocto-trees/sources/poky... 
[...]
Building target hytem-banana-zeus
[...]
NOTE: Tasks Summary: Attempted 5227 tasks of which 4790 didn't need to be rerun and all succeeded.
FATAL: build for hytem-banana-zeus failed 'Namespace' object has no attribute 'sdk'
```

pboettch, 2020-03-30:
This is because the cook-command does not have an sdk-flag in its command line arguments.

cpb-, 2020-03-30:
We could add the `--sdk` option for the `cook` command.
Are you ok with this?

pboettch, 2020-03-30:
Another option would be that the sdk-option is always selected if cook is used. Cook is a high-level-command, options should not be redundant. At least not too many of them.

Hmm. I don't have any real answer.

cpb-, 2020-03-30:
You're right, always selecting `sdk-option` for `cook` command makes more sense than adding an option.

commit 85ef8d3f (2020-03-31): [DEV] The `cook` command builds the sdk.

Fixes: #29
closed 2020-03-31 by cpb-