cpb-/yocto-cooker #32: `chef cook` failure

cpb-, 2020-03-31:
Calling `chef cook <menu>` fails if `chef init <menu>` was not called first:

```
[Yocto-trees@why-cpb]$ chef cook custom-systems.menu 
Update layers in project directory
Updating source /home/testing/Build/Yocto-trees/sources/poky... 
Updating source /home/testing/Build/Yocto-trees/sources/meta-openembedded... 
Updating source /home/testing/Build/Yocto-trees/sources/meta-sunxi... 
Updating source /home/testing/Build/Yocto-trees/sources/meta-qt5... 
Generating build dirs for target in project directory
Traceback (most recent call last):
  File "/home/cpb/bin/chef", line 460, in <module>
    ChefCall()
  File "/home/cpb/bin/chef", line 213, in __init__
    self.clargs.func() # call function of selected command
  File "/home/cpb/bin/chef", line 232, in cook
    self.prepare()
  File "/home/cpb/bin/chef", line 262, in prepare
    self.generate()
  File "/home/cpb/bin/chef", line 244, in generate
    self.prepare_build_directory(directory, content)
  File "/home/cpb/bin/chef", line 359, in prepare_build_directory
    dl_dir = '${TOPDIR}/' + os.path.relpath(self.config.dl_dir(), path)
  File "/home/cpb/bin/chef", line 95, in dl_dir
    return os.path.join(self.project_root(), self.cfg['dl-dir'])
KeyError: 'dl-dir'
```

pboettch, 2020-03-31:
Yes, with my dl-dir-fix you need to re-issue a chef init -f in an existing project-dir to have the dl-dir set. Be careful of default values.

cpb-, 2020-03-31:
I tried to remove the `.chefconfig` file, but it's not better:

```
$ chef cook custom-systems.menu 
Update layers in project directory
Traceback (most recent call last):
  File "/home/cpb/bin/chef", line 460, in <module>
    ChefCall()
  File "/home/cpb/bin/chef", line 213, in __init__
    self.clargs.func() # call function of selected command
  File "/home/cpb/bin/chef", line 232, in cook
    self.prepare()
  File "/home/cpb/bin/chef", line 261, in prepare
    self.update()
  File "/home/cpb/bin/chef", line 254, in update
    self.download_source(source)
  File "/home/cpb/bin/chef", line 290, in download_source
    local_dir = os.path.realpath(self.config.layer_dir(local_dir))
  File "/home/cpb/bin/chef", line 91, in layer_dir
    return os.path.join(self.project_root(), self.cfg['layer-dir'], name)
KeyError: 'layer-dir'
```

I thought that `chef cook <menu>` is the upper-level command that could run without `.chefconfig` file, using defaults values. 

`chef init` is a more advanced command, that user doesn't need to know to just try `chef`.

pboettch, 2020-03-31:
upper level commands are based on lower level commands and lower level commands use the (loaded, just created) config. Upper level commands have to call them correctly with the arguments. 

Hmm...

pboettch, 2020-03-31:
We need a test environment so see that the addition/changes of commands do not break other commands.

commit ebd77341 (2020-03-31): [DEV] Default config before reading `.chefconfig`

Initialize the attributes of the `Config` instance with default values before attempting to read the configuration file.

The subdirectories are:

- `sources` for the layers
- `builds` for the build directories
- `download` to store downloaded objects.

Fixes: #32
closed 2020-03-31 by cpb-