cpb-/yocto-cooker #34: init and then prepare, json menu needed as param ?

adri1mart1, 2020-04-06:
Hello there,

I am in the following flow case where I would like a specific DL_DIR so I use:
```
chef init -d dl_dir menu.json
```

Then I am using 

```
chef prepare menu.json
```

What's the point of setting menu.json as parameter in prepare command if I already init my repository (So the menu is already in the .chefConfig file) ?

pboettch, 2020-04-06:
If you use `init` (a low-level command) then use `update` and  `generate` to download and generate.

Prepare is actually `init`+ `update` + `generate`.

cpb-, 2020-04-06:
It may be simpler if the directories options (`-b`, `-l`, `-d`) could be used at each level (`init`, `prepare`, `cook`), no?

cpb-, 2020-04-06:
FYI my workflow is `chef cook my-menu.json` for the first run, then only `chef build`.

Sometimes I need to call `bitbake -c clean this-package`. So I source `poky/oe-init-build-env` first. It would be nice to have a high-level `chef clean this-package`.

adri1mart1, 2020-04-06:
I see multiple ways of solving this, from an external point of view, the command line flow is not so easy to guess, especially if you, like me, use the low level functionalities. 
I would at least clarify the command line argument chapter in the doc by adding a paragraph to prepare the fast way (chef cook / chef build), and a detailled one to explain (chef init / update / generate / build). 
Saying that I don't know which chapter should be the "prepare" keyword

pboettch, 2020-04-06:
> It may be simpler if the directories options (`-b`, `-l`, `-d`) could be used at each level (`init`, `prepare`, `cook`), no?

I don't like it. Adding the low-level arguments to the higher level ones will confuse. 

Compare it with `git clone`, it is actually a combination of git init, fetch, config, checkout and other. Still clone does not have all the arguments for its lower-level commands.
I see it the same way here.

Maybe `prepare` should be removed and `cook` would be verbose printing out the low-level commands it calls. One by one.

cpb-, 2020-04-06:
> Maybe `prepare` should be removed and `cook` would be verbose printing out the low-level commands it calls. One by one.

Sounds good to me.

cpb-, 2020-04-07:
One more thing about chef options.

What about a `chef cook --no-sdk menu.json` option to speed up the first run?

adri1mart1, 2020-04-10:
hum I think maybe it could be set in local.conf instead no ?

pboettch, 2020-04-15:
Prepare is now removed via #45.

closed 2020-04-15 by pboettch