cpb-/yocto-cooker #35: add some command line tooltips

adri1mart1, 2020-04-06:
Hello there,

I have a menu.json that when prepared by chef, I reach the following tree organisation:

```bash
.
├── builds
│   └── build-my-image
└── layers
    ├── meta-openembedded
    ├── meta-xxx
    ├── poky
    └── meta-yyy
```

Chef is very handy to build the whole image but let's say i simply want the bitbake sourced environnement. If I run the classic:

```bash
source layers/poky/oe-init-build-env
```
I get the following tree struct as results:
```bash
.
├── build
│   └── conf
├── builds
│   └── build-msdi-image
└── layers
    ├── meta-openembedded
    ├── meta-xxx
    ├── poky
    └── meta-yyy
```

I would love to have more written info when running chef prepare, something like:
```bash
chef prepare menu.json 
Update layers in project directory
Downloading source from  git://git.yoctoproject.org/poky
Updating source ....
[...]

Environment ready to build images:
Run chef build to compile everything 
or 
source the bitbake environnement using source xxx xxx 
Generating build dirs for target in project directory
```

What do you think ?

cpb-, 2020-04-06:
Nice idea. I'm a huge fan.

pboettch, 2020-04-06:
That was the idea of the target-info command.

cpb-, 2020-09-08:
The `show` command implements this feature.

closed 2020-09-08 by cpb-