cpb-/yocto-cooker #61: New sub-command?

cpb-, 2020-06-09:
I think an interesting new sub-command of `chef show` could be a whole list of all the command lines needed to perform the same task as `chef cook`.

For example it could dump a kind of shell script like:

```
#! /bin/sh
mkdir sources
cd sources
git clone git://git.yoctoproject.org/poky -b zeus
cd poky
git checkout zeus-22.0.2
cd ../
[...]
cd builds
. sources/poky/oe-init-build-env build-my-image
cat > build-my-image/conf/local.conf <<-EOF
 [...]
EOF
bitbake my-target
bitbake -c populate_sdk my-target
```

So the long time bitbake user could check what `chef` does behind the scenes.
And it would help to convince new users of the interest of `chef`.

pboettch, 2020-06-09:
I think it would be better fitted with `-n` command line switch. And then channeling all commands via a function which only prints out the command instead of executing them.

```bash
chef -n cook <menu>
```

would show all commands executed or 

```bash
chef -n build <build-config>
```

as another example how to use `-n`.

cpb-, 2020-09-08:
Closed by 0e4f5e9

closed 2020-09-08 by cpb-