cpb-/yocto-cooker #74: Cloning ssh without "dir" results in a bad naming

Louson, 2020-12-04:
Cloning with ssh without the variable "dir" will have a bad directory name.

```
{ "url": "git@gitlab.com:foo/bar.git" }
```

will be downloded in a directory named: 'it@gitlab.com:foo/bar.git'
while it should probably be 'bar'

Problem comes from cooker.py:145
```
local_dir = url.path[1:]
```

pboettch, 2020-12-04:
Could catch. Support for "short" URLs was added recently and of course without a test. (currentlty there are no tests which are cloning anything, though).

Louson, 2020-12-04:
Actually, if we don't gie a dir name to https link, the result may not be what expected. 

eg: https://foo/meta-bar
will be placed in layers/foo/meta-bar
instead of (probably) layers/meta-bar

pboettch, 2020-12-04:
No. It turned out to be useful to have a sub-directory for layers following their github.com-path.

So `layers/foo/meta-bar` is the expected result when not specifying `dir`.

commit db84c4f0 (2021-06-09): sources: fix path extraction from short git-URL (#84)

Fixes #74

Basically this URL `git@github.com:path/to/repo` now
gives the right path.
closed 2021-06-09 by cpb-