cpb-/yocto-cooker #81: build-dir not created even if inherited build contains "target" attribute

jb-flamant, 2021-05-29:
Hi,

I tried to use the menu inheritances system and it didn't work "like i wanted to". It's because of the [**target** attribute requirement ](https://github.com/cpb-/yocto-cooker/blob/1d4cae7522f012720b292165445d31686ab3a1c3/cooker/cooker.py#L410).

I'd like to know why is there such a restriction. I would like to use something like this :

```json   
    ...
     "builds" : {
                "pi-prod": {

                        "target" : "image-prod",

                        "layers" : [
                                "meta-raspberrypi"
                        ]
                },
                "pi2": {

                        "inherit" : [ "pi-prod" ],

                        "local.conf": [
                                "MACHINE                    = 'raspberrypi2'   "
                        ]

                },
                "pi3": {

                        "inherit" : [ "pi-prod" ],

                        "local.conf" : [
                                "MACHINE                    = 'raspberrypi3'   "
                        ]
                },
                ...
        }
        ...
```
and get **build-dir** for **pi2** and **pi3** but not **pi-prod**.

Thank you.

PS : yocto-cooker is really a nice tool !

cpb-, 2021-05-30:
Hi JB,

Thanks for your comment.

The `target` attribute was chosen to distinguish between buildable and not-buildable configurations.

But your remark is interesting, and maybe we could use the combination `target` + `MACHINE` to declare a configuration as buildable and let `target` be inherited ?
In this case, we would have to parse the `local.conf` array to check for `MACHINE` presence.

RomainNaour, 2021-05-31:
Hello Christophe,
What about making buildable and not-buildable configurations explicit ?
Like on gitlab-ci job name starting with a '.' are template (ignored job) ?

RomainNaour, 2021-06-01:
@jb-flamant  see #82

jb-flamant, 2021-06-01:
> Hi JB,
>
> Thanks for your comment.
>
> The `target` attribute was chosen to distinguish between buildable and not-buildable configurations.
>
> But your remark is interesting, and maybe we could use the [...]

Hi Christophe,

Thanks for your quick reply. I think @RomainNaour solution is a good one. It permits a lot of combination and is maybe simpler.

cpb-, 2021-06-03:
Hi @jb-flamant,
If you're Ok with the solution proposed above, can you close this issue.
Thanks

closed 2021-06-03 by jb-flamant