cpb-/yocto-cooker #99: `cooker shell` failure

cpb-, 2021-11-25:
When running `$ cooker shell <build-name>`, I got the following error:

 ```
Error: /home/user/oe-init-build-env doesn't exist!
Please run this script in oe-init-build-env's directory.
```

Other commands (`show`, `cook`, etc.) work as expected.

`cooker --version` gives `# 1.1.0`

pboettch, 2021-11-25:
Are you sure you're not using your dev-branch which implements the support for different openembedded-distributions?

cpb-, 2021-11-30:
When `/bin/sh` is a link to `/bin/bash` the `shell` sub-command runs as expected.
The problem arises when `/bin/sh` is a link to `/bin/dash`.
This was the case in my build container above.

Running manually the command that `cooker` tries to run gives the following results

With Bash:

```
$ /bin/bash -c ". /home/user/layers/poky/oe-init-build-env my-build; /bin/bash"

### Shell environment set up for builds. ###

You can now run 'bitbake <target>'
[...]

$ which bitbake
/home/user/layers/poky/bitbake/bin/bitbake

$ exit
exit
```

With Dash:

```
$ /bin/dash -c ". /home/user/layers/poky/oe-init-build-env my-build; /bin/dash"
Error: /home/user/oe-init-build-env doesn't exist!
Please run this script in oe-init-build-env's directory.
```

The easier way to resolve this could be a direct call to `/bin/bash`.
Note that we already have bash-dependencies. For example line 580 of current `cooker.py` version;

```
580      command_line = 'env bash -c "source {} {} && bitbake {} {}"'.format(init_script, directory, bb_task, bb_target)
```

So the question is: do we need to fight to remove Bash dependencies, or could we just replace 

```
588      shell = os.environ.get('SHELL', '/bin/sh')
```

by 

```
588      shell = os.environ.get('SHELL', '/bin/bash')
```

Any suggestion?

pboettch, 2021-12-01:
Can't we use python-function to do system-calls? Without bothering which shell is default?

cpb-, 2021-12-01:
Not sure if we can source a shell script (`oe-init-build-env` for example) directly with python functions.

RomainNaour, 2021-12-16:
> When `/bin/sh` is a link to `/bin/bash` the `shell` sub-command runs as expected. The problem arises when `/bin/sh` is a link to `/bin/dash`. This was the case in my build container above.

At least with Dunfell, dash is still an issue with Yocto and you have to reconfigure dash to use bash (dpkg-reconfigure dash).
I had the issue recently.

PR #108 (2022-02-18): [DEV] Default to `bash` when running `cooker shell`.
When the `SHELL` environment variable is not defined, we have
to default to `/bin/bash` (like when we launch `bitbake`) instead
of `/bin/sh` which is often a link to `/bin/dash`.

Closes: #99
commit 6a9d122e (2022-02-18): [DEV] Default to `bash` when running `cooker shell`. (#108)

When the `SHELL` environment variable is not defined, we have
to default to `/bin/bash` (like when we launch `bitbake`) instead
of `/bin/sh` which is often a link to `/bin/dash`.

Closes: #99
closed 2022-02-18 by cpb-