cpb-/yocto-cooker #103: Support multiple targets for one build

Louson, 2022-01-21:
What about supporting multiple targets for one build ?

The menu could have an entry `targets` (just as `target`) with an array.

The BuildConfiguration class would have a list of targets

bitbake would be executed for each target

Louson, 2022-01-21:
This is a enhancement suggestion. I may work on it if ever it is a good idea

cpb-, 2022-01-21:
This seems to be a really good idea...

cpb-, 2022-01-21:
Hmm, `target` is the name of the image argument for `bitbake`. In fact I rarely use multiple targets for a project.

But I often build the same image for several machines.
How about adding a `machines` attributes to the menu,  and running `bitbake` successively after updating `MACHINE="..."` line in `local.conf` ?

Louson, 2022-01-21:
But the build can be very dependent on your machine. While the image use the same build.

Suppose you have two images for a hardware, or you just want to build the kernel and u-boot.

cpb-, 2022-01-21:
Ok, the `targets` idea is interesting.

I'll submit a different RFC for the `machines` entry idea. In my use case it could be useful because I have some projects where the same image is built for different machines. All the hardware-specific parameters are stored in the machines files.

pboettch, 2022-01-23:
Multiple targets, we thought about it in the beginning and as we were limiting us to image-targets we said it is unlikely to have multiple of them.
The goal of a menu is to ultimately also automate the build which you will do regularly. How often do you build some sub-targets automatically (which are not build by the image? For development I usually enter a `cooker shell` and run bitbake on the sub-target manually.

If sub-targets or multiple images per cooker-build are really necessary. The target-property can be turned into an array while still accepting a string. (schema-wise there is no problem). I'd prefer to not have a `target` and `targets` property. Maybe we could obsolete target in favor of targets.

Louson, 2022-01-24:
Ok, I'll try to suggest some patch

pboettch, 2022-02-14:
Closed via #106

closed 2022-02-14 by pboettch