cpb-/yocto-cooker #109: Updating "rev" of a source in menu does not trigger a new download

fmalorey, 2022-03-09:
When updating "rev" param of a git source in my menu.json, a `cooker cook menu.json` command doesn't trigger a new download.

This leads to an error on Udpate step ie: "FATAL: Unable to checkout rev".

Workaround found is to manually layer directory to force a new download.

cpb-, 2022-03-10:
I tried the following procedure to reproduce the problem, but it doesn't appear. 

## Step 1

Create a dedicated public repo on Github.

Here I choose 'will-be-deleted' with a default README

Clone it in a directory independant from Cooker working space.

```
$ git clone https://github.com/cpb-/will-be-deleted
```

## Step 2

Modify the `README.md` and commit it. Create a first tag and push it.

```
$ echo 1.0 >> README.md
$ git commit README.md -m 1.0
$ git tag 1.0
$ git push
$ git push --tags`
```

## Step 3

Create a simple menu named `test-menu.json` like this:

```
{
  "sources" : [
    { "url": "https://github.com/cpb-/will-be-deleted.git", "dir": "test-layer", "rev": "1.0" }
  ],

  "layers":  [ "will-be-deleted" ],

  "builds" : {
    "test-build": {
      "target" : "test-target"
    }
  }
}
```

## Step 4

Initialize Cooker and do a first `fetch`.

```
$ rm -f .cookerconfig 
$ cooker init test-menu.json
$ cooker update
```

## Step 5

Check the `README.md`.

```
$ cat layers/test-layer/README.md 
# will-be-deleted

1.0
```

## Step 6

In the outside repository update the README, add a tag and push.

```
$ echo "2.0" >> README.md 
$ git commit README.md -m 'update'
$ git tag 2.0
$ git push && git push --tags
```

## Step 7

Edit Cooker menu and update the tag revision.

```
 [...]
    { "url": "https://github.com/cpb-/will-be-deleted.git", "dir": "test-layer", "rev": "2.0" }
 [...]
```

## Step 8

Rerun Cooker to fetch the repo.

```
$ cooker update
```

## Step 9

Check if an update has been done:

```
$ cat layers/test-layer/README.md 
# will-be-deleted

1.0
2.0
```

pboettch, 2022-03-11:
This is a reason why I don't often use cook. Maybe we could create a better error message around this problem.

cpb-, 2022-03-14:
There's an upcoming pull request that displays `git` error in case of failure (indicating that the layer has been modified and that the local changes prevent the update).
Maybe we could also had a `--drop` command-line option for `cooker cook` / `cooker update` that discard any local modifications ?

commit f6f0f049 (2022-03-14): [DEV] Replace `os.system()` calls by `subprocess.run()`.

To display Git error messages in case of problem.

Closes: #109
closed 2022-03-14 by cpb-