cpb-/yocto-cooker #117: Error when fetching a repository

cpb-, 2022-05-10:
Here's an error I encountered this morning while running cooker . The error seems to be related with Poky fetching.

```
$ cooker cook meta-ily/ily-training.json 
# Update layers in project directory
# Downloading source from  git://git.yoctoproject.org/poky
Traceback (most recent call last):
  File "/usr/local/bin/cooker", line 33, in <module>
    sys.exit(load_entry_point('cooker==1.3.0', 'console_scripts', 'cooker')())
  File "/usr/local/lib/python3.6/site-packages/cooker/cooker.py", line 902, in main
    CookerCall()
  File "/usr/local/lib/python3.6/site-packages/cooker/cooker.py", line 834, in __init__
    self.clargs.func()  # call function of selected command
  File "/usr/local/lib/python3.6/site-packages/cooker/cooker.py", line 861, in cook
    self.commands.update()
  File "/usr/local/lib/python3.6/site-packages/cooker/cooker.py", line 369, in update
    self.update_source(source)
  File "/usr/local/lib/python3.6/site-packages/cooker/cooker.py", line 402, in update_source
    self.update_directory_initial(method, remote_dir, local_dir)
  File "/usr/local/lib/python3.6/site-packages/cooker/cooker.py", line 420, in update_directory_initial
    complete = CookerCall.os.subprocess_run(["git", "clone", "--recurse-submodules", remote_dir, local_dir], None)
  File "/usr/local/lib/python3.6/site-packages/cooker/cooker.py", line 63, in subprocess_run
    return subprocess.run(args, capture_output=capture_output, cwd=cwd)
  File "/usr/lib64/python3.6/subprocess.py", line 423, in run
    with Popen(*popenargs, **kwargs) as process:
TypeError: __init__() got an unexpected keyword argument 'capture_output'
```

pboettch, 2022-05-10:
version python?

pboettch, 2022-05-10:
capture_output was added in python 3.7. Probably you are using 3.6

closed 2023-05-22 by cpb-