cpb-/yocto-cooker #129: Ajouter un mot de passe à root

Amnesix, 2022-10-02:
Bonjour,

Je viens de tenter d'ajouter un mot de passe à root en ajoutant la ligne suivante dans "local.conf" :
"EXTRA_USER_PARAMS:append = 'usermod -p \$5\$NQL5JlW1yrj/rgji\$AdRG9oSdsIT.PH./ZwJG0GQrsLqDovb7.FtvcDoWhu6 root;'"

Problème en executant cooker cook :
FATAL: menu load error: Invalid \escape: line 23 column 50 (char 697)

Si je supprime les backslash, la compilation arrive au bout mais le mot de passe n'est pas reconnu, ce qui s'explique probablement par le fait que le local.conf contient les $ et non des \$ ?

cpb-, 2022-10-02:
Bonjour Jean,

Pour inscrire dans un menu une ligne `EXTRA_USERS_PARAMS`, il faut mettre des doubles _backslashes_ devant les `$` ainsi :

```
 "EXTRA_USERS_PARAMS:append = 'usermod -p '\\$5\\$NQL5JlW1yrj/rgji\\$AdRG9oSdsIT.PH./ZwJG0GQrsLqDovb7.FtvcDoWhu6 root;'"
 ```

Amnesix, 2022-10-03:
Wow ! C'était pourant évident ! Merci.

closed 2022-10-04 by cpb-