cpb-/yocto-cooker #144: Will not cook in CROPS

chrisfair, 2023-06-05:
This may be out of scope, and if it is I am sorry for reporting it.   I am attempting to use this library from within a docker container (a modified CROPS image from the megamanual...I just thought it would be nice to have this incorporated in my image).   I successfully create the init file but I get the following error whenever I attempt to build or update....

Traceback (most recent call last):
  File "/usr/local/bin/cooker", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/cooker/cooker.py", line 1023, in main
    CookerCall()
  File "/usr/local/lib/python3.6/dist-packages/cooker/cooker.py", line 955, in __init__
    self.clargs.func()  # call function of selected command
  File "/usr/local/lib/python3.6/dist-packages/cooker/cooker.py", line 982, in cook
    self.commands.update()
  File "/usr/local/lib/python3.6/dist-packages/cooker/cooker.py", line 417, in update
    self.update_source(source)
  File "/usr/local/lib/python3.6/dist-packages/cooker/cooker.py", line 453, in update_source
    self.update_directory_initial(method, local_dir, remote_dir, branch, rev)
  File "/usr/local/lib/python3.6/dist-packages/cooker/cooker.py", line 467, in update_directory_initial
    complete = CookerCall.os.subprocess_run(["git", "ls-remote", remote_dir ], None)
  File "/usr/local/lib/python3.6/dist-packages/cooker/cooker.py", line 64, in subprocess_run
    return subprocess.run(args, capture_output=capture_output, cwd=cwd)
  File "/usr/lib/python3.6/subprocess.py", line 423, in run
    with Popen(*popenargs, **kwargs) as process:
TypeError: __init__() got an unexpected keyword argument 'capture_output'

The docker container has python36 in it....I satisfied the requirements as far as I could tell in the setup.py, are there any suggestions on how to get this to work.   I was about to write myself a build tool in bash when I found cooker and thought I would give it a go before I write anything.   If this is in scope then my bug is that I cannot build and get the preceeding error whenever I attempt to cook within a CROPS container.

Thanks!
Chris

cpb-, 2023-06-06:
Hi Chris,

The `capture_output` parameter of `subprocess_run()` was introduced in Python 3.7.
I suggest to use tag `1.3.0` or `1.2.0` of Cooker, and not `master` branch head.

:-)

chrisfair, 2023-06-14:
.Hi sorry I did not get back to you guys, that was my problem and yup....worked great.   BTW...works really well cooupled with CROPS...almost turn key

closed 2023-06-15 by cpb-