cpb-/yocto-cooker #152: [RFC] How about `dumpenv` command?

cpb-, 2024-05-30:
I would be interested in a `cooker dumpenv [build]` command.

Instead of running `bitbake <image-name>` it would run `bitbake -e <image-name>` and redirect the standard output to a `environment.txt` file located in the build directory.

It would be useful to see the content of some quite complex variables (`*OVERRIDES` or `*_FEATURES` for example) after `bitbake` parsing of the recipes.

Any comment?

boucman, 2024-05-30:
hmm

bitbake -e has multiple variants, all of which are usefull

bitbake -e
bitbake -e <image>
bitbake -e <any recipe>

how would you proposal work for that ?

cpb-, 2024-05-30:
🤨

pboettch, 2024-05-30:
what about `cooker shell <target> bitbake -e` ?

cpb-, 2024-05-31:
```
$ cooker shell raspberrypi4   bitbake -e
usage: cooker [-h] [--debug] [--version] [-v] [-n]
              {cook,init,update,diff,log,generate,show,build,shell,clean} ...
cooker: error: unrecognized arguments: -e

$ cooker shell raspberrypi4   "bitbake -e"
/bin/bash: line 1: bitbake -e: command not found
```

pboettch, 2024-06-02:
Doesn't work, but would this be an acceptable solution?

cpb-, 2024-06-15:
Never mind, I've done it another way.

closed 2024-06-15 by cpb-